atc1441/ATC_RF03_Ring #1: smart ring with buttons and LED

x3e, 2024-06-19:
When searching for a place to buy this, I came across a slightly different smart ring that has capacitive touch points and an outward-facing led (at least in the product video) Do you think the SoC wil be the same?

A downside of this one is that the epoxy layer on the inside is not transparent and I wonder if it will be possible to wire up the swd like you did.

Anyway, I ordered one and if you want I 'll report back once I have received it.

atc1441, 2024-06-19:
Hey, 

that ring looks like it is just emulating a mouse. There is a chance that the firmware in it will be fixed.

It seems its missing any additional features like Accl or Heartrate sensor.
Still interesting.

Looking forward to a report from you with hopefully a name of the chip used :) 

[image]

x3e, 2024-07-03:
Seems like this ring is quite different. It has a different construction as well, with the metal ring being on the inside and not on the outside of the ring. Disassembly seems easier, although I probably broke a trace when peeling off the PCB (heating the glue might help).

Sadly, the IC is different. I couldn't get a good photograph of the marking, but it is a JL IC marked with something like "BPOP203-23A2". Couldn't find any info on it.

If you want, you can close this issue as it doesn't seem to be related to your work on the other ring.

[image]
[image]
[image]

atc1441, 2024-07-03:
Thank you for the insights of the ring,

The SoC is this one: https://www.lcsc.com/product-detail/Microcontroller-Units-MCUs-MPUs-SOCs_JieLi-Tech-AC6323A4_C5440136.html

Quite capable but with a bad sleep current. There would be an SDK but as you mention its a different Project overall.

closed 2024-07-03 by atc1441