atc1441/ATC_RF03_Ring #4: Sdk compilation

decaduto, 2024-06-21:
Hi,

I was analyzing by my own those smart rings too, One question: have you already found a way for compiling the firmware from the SDK?

Thank you

decaduto, 2024-06-21:
PS: in the SDK there isn't any makefile and in general the few docs I found are in cinese, but I haven't found any instruction for the build processo.

atc1441, 2024-06-21:
You can use Keil for that in the stock implemenation right now

osresearch, 2024-06-21:
The SDK appears to have a [license that allows source redistribution](https://github.com/osresearch/bx3-sdk/tree/main?tab=License-1-ov-file#readme), so I've created a standalone tree [osresearch/bx3-sdk](https://github.com/osresearch/bx3-sdk/) cloned from https://gitee.com/BXMicro/SDK3 . diffing the trees shows that this is the same as the 20 part tar file containing a zip file of the 3.3.6 release stored in this repo.

atc1441, 2024-06-21:
Thanks for that @osresearch yes that V3.3.6 in this repo is just a gitee clone to give easy access to it.

its worth to note that the older V2 SDK has more infos and tools inside which is missing on the newer SDK version

decaduto, 2024-06-22:
> You can use Keil for that in the stock implemenation right now

The only makefile present in the v3 SDK used 'gcc' as a default compiler and I've noticed that the names of the paths are wrong (gnu instead of GNU etc..)
The most important things to find RN Is a build script, Is It present from the V2?

decaduto, 2024-06-22:
> Thanks for that @osresearch yes that V3.3.6 in this repo is just a gitee clone to give easy access to it.
> 
> its worth to note that the older V2 SDK has more infos and tools inside which is missing on the newer SDK version

Also, do you have some idea if the 'demo' folder contains the effettive code used in those Rings firmware? If I find some time I try to reverse the ota Fw and search for common patterns for a specific 'demo' code.

atc1441, 2024-06-23:
Here is a manual to set up eclipse for the SDK (untested but looks good)
https://www.wpgdadatong.com/blog/detail/44133

This way a custom setup is simpler to get running

decaduto, 2024-06-23:
> Here is a manual to set up eclipse for the SDK (untested but looks good) https://www.wpgdadatong.com/blog/detail/44133
> 
> This way a custom setup is simpler to get running

I made It working for Linux, at least I fixed the problem with Cmake, the only problem Is that I don't know how to get fromelf, which seems to be offerend by the armcc compiler tools.

Also, I analyzed the firmware from the OTA server, It has a lot of strings which aren't present in the source code, I can Imagine that this Is due to  OEM modifications.

osresearch, 2024-06-23:
I've also been hacking on a `Makefile` to work with the existing demos and have things compiling up to the link stage. Are some of the functions stored in (an actual?) ROM? There is no source for `memset()` or `ke_timer_set()` in the source tree, and instead they are referenced in `components/bluex/ble/controller/rom_syms_armcc.txt` as if they were located at constant offsets.

For Magic Lantern back in 2010 we maintained a similar table for each ROM version (such as [5D Mark 2 v2.1.2](https://foss.heptapod.net/magic-lantern/magic-lantern/-/blob/branch/unified/platform/5D2.212/stubs.S?ref_type=heads) and had some compile time magic to generate these symbols at the correct address.  Is there a better way these days?

atc1441, 2024-06-23:
I know of such thinks still being used on other chips. so not unlikely

decaduto, 2024-06-23:
> I've also been hacking on a `Makefile` to work with the existing demos and have things compiling up to the link stage. Are some of the functions stored in (an actual?) ROM? There is no source for [...]

Yes, in my experience I found a lot of glibc basic functions in ROM, like memcpy, strncpy and memset too, but honestly I think It would be Better to create a stub which contains inline functions for memset and so on, It should be pretty fast to do.

Have you worked with the sdk3 or the sdk2.1? i am able to compile the sdk2.1, but It fails because It cannot find fromelf, which Is inside the keil compiler, but I haven't a Copy of It, do you have?

decaduto, 2024-06-23:
guys, some additional analysis on the firmware "R02_1.00.05_211220.bin": I don't think the chip is the RF03, instead, I think is somewhat linked to a realtek 87xx model, I have searched for some matching strings in the SDK, and there is almost NOTHING in common, but, if you try to do a code search on github, you will came across a LOT of strings used by realtek SDK, some examples follows: https://github.com/GeFengthe/RTL8762/blob/f80ae6a2362d986a3d69c764ca1cda3ec39155c5/src/mcu/rtl876x/system_rtl8762c.c#L837  https://github.com/ART-Badge/SDK/blob/116b2296a9f6442a85c68daf4855d9ed36cc330d/bsp_for_art-badge/libraries/src/app/wristband/sensor_hub/hub_charger.c#L53  https://github.com/ART-Badge/SDK/blob/116b2296a9f6442a85c68daf4855d9ed36cc330d/bsp_for_art-badge/libraries/src/app/wristband/sensor_hub/hub_hrm.c#L4  https://github.com/GeFengthe/RTL8762/blob/f80ae6a2362d986a3d69c764ca1cda3ec39155c5/src/mcu/rtl876x/system_rtl8762c.c#L378  https://github.com/Librianets/RTL8763BF/blob/c1df8b06940ae2f25d3bde31d3d6ebef662dfff5/SDK/1.0.2/inc/bluetooth/gap/gap_lib/gap_vendor.h#L172   https://github.com/Jieli-Tech/fw-AC63_BT_SDK/blob/97114708aa476b5f30eebbe5877bda91f187bcb6/apps/common/third_party_profile/common/custom_cfg.c#L1283  https://github.com/ART-Badge/SDK/blob/116b2296a9f6442a85c68daf4855d9ed36cc330d/bsp_for_art-badge/libraries/src/platform/dfu_flash.c#L981  https://github.com/ambiot/ambd_sdk/blob/53ab0f0272d2b045f1e697bfd3d87df7ab7d133f/component/common/bluetooth/realtek/sdk/src/ble/profile/client/ancs_client.c#L546 (sorry for the confusion, I have only some minutes rn), the question is: are we sure that the SDK is the correct one? and what if the chipset for this specific firmware is the RTL87xx? I haven't checked the other 2 firmwares, it may be possible that they choose to use both realtek and Bluex mcu, additional investigation is needed imho.

decaduto, 2024-06-23:
I confirm that only the second OTA Fw Is effectively linked with the rtl8762 mcu, It has a different size compared to the other 2 and presents a completely different strings set, I think we should fork the development in multiple directions.

atc1441, 2024-06-24:
Take a look at the firmware dump I got out of the RF03 itself, everything fits for it to be the BlueX one, "BK24" header memory locations etc. (BK2400 is the "top" name of all the RF01-RF08 models)

The Colmi company also makes a lot of Smartwatches with the RTL SoC's so they will be using one codebase as also the BLE protocol is the one used in the Smartwatches.

They will just have ported the existing code fitting to the RF03.
Also the FW ota header fits into the SDK: 
https://github.com/osresearch/bx3-sdk/blob/main/components/bluex/ble/profiles/bxotas.c#L143

So unless BlueX and RTL is one company under the same name I would not say it is any RTL Chip

I can be totaly wrong of course!

atc1441, 2024-06-24:
So yeah, just to confirm, the running firmware is definitely coming from the BlueX SDK :)

As an example take a look at the decompiled function 0x80E790 (OTA firmware plus 0x803000 offset)
It is clearly the I2C_scl init function. And looking at its calling function(0x811BC0) you will also see the other GPIO inits.
Also to get the same start, 0x8139F0 is main.

https://github.com/osresearch/bx3-sdk/blob/main/platform/bluex/apollo/apollo00/driver_service/bxd_iic.c#L235

[image]

osresearch, 2024-06-24:
I see lots of common strings from the BX3 SDK in the OTA firmwares, as well.  Things like `"SDK3"`, `rf01_onchip` and `user_scan_rsp_data` appear in the sdk3 tree, and the flash partition magic `01PE` followed by different sections like `bl`, `app`, and `kvdb`.

I have the `examples/base/app_base` compiling up to the linking stage with a hand-hacked makefile and linker script (and the `ble_init()` call commented out), but need to setup for reprogramming via the hardware pins so that I can recover from a botched flash.  Unfortunately the assembly `.s` files are in a non-gcc dialect and will need some massaging to work with `arm-none-eabi-as`.  The BLE stack uses them, which is why I've temporarily removed it from my build.

decaduto, 2024-06-24:
> I see lots of common strings from the BX3 SDK in the OTA firmwares, as well. Things like `"SDK3"`, `rf01_onchip` and `user_scan_rsp_data` appear in the sdk3 tree, and the flash partition magic [...]

I am talking only for the R02_1.00.05_211220.bin firmware (the One with 274 KB of size), the other 2 are from the SDK.

osresearch, 2024-06-24:
Sorry, I had missed that detail! Yes, the R02 OTA firmware is totally different from the other two and appears to be for a totally different product. Maybe some sort of wristband with a display?

decaduto, 2024-06-24:
> So yeah, just to confirm, the running firmware is definitely coming from the BlueX SDK :)
> 
> As an example take a look at the decompiled function 0x80E790 (OTA firmware plus 0x803000 offset) It is clearly the I2C_scl init function. And looking at its calling function(0x811BC0) you will also see the other GPIO inits. Also to get the same start, 0x8139F0 is main.
> 
> https://github.com/osresearch/bx3-sdk/blob/main/platform/bluex/apollo/apollo00/driver_service/bxd_iic.c#L235
> 
> ![Decompiled](https://private-user-images.githubusercontent.com/1102694/342180375-d41fb421-8dce-4ff7-a4f7-6eea69d572b2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyMTI2MjQsIm5iZiI6MTcxOTIxMjMyNCwicGF0aCI6Ii8xMTAyNjk0LzM0MjE4MDM3NS1kNDFmYjQyMS04ZGNlLTRmZjctYTRmNy02ZWVhNjlkNTcyYjIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MjRUMDY1ODQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzM5ZDNhNzQ2MDNjNWIyMThiNGRmYzgxNDcwODQ5NWUzZjE2NzFhNTQzODEwZjVmNjUwMWY1N2RhYWM4M2RhYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.u0lrQcYDdpGYXu4T-OA6uZi8E_vhDcY_A4TN4EjOGEk)

Well, I cannot find that Function in R02_1.00.05_211220.bin (I am using ghidra), I have searched Also for the scalar values but there isn't anything related, also note that there isn't any strings like "rf02" or "bluex" in that binary, I suppose that each ring could have only one of the 3 firmwares you uploaded in this repo, Is that correct?

For the header, they could have adapted It to fit with the other 2 models which uses the same bx3 SDK, I cannot believe that the R02_1.00.05_211220.bin Is based on the sdk3, it has too many things which don't fit with the other 2 firmwares.

Also note that the STK8321 "driver" used in the Ximax R02 smart ring isn't referenced in the bluex SDK but Is present in the rtl8762 SDK.

decaduto, 2024-06-24:
> Sorry, I had missed that detail! Yes, the R02 OTA firmware is totally different from the other two and appears to be for a totally different product. Maybe some sort of wristband with a display?

I linked above the possible SDK, i think Is based on a rtl8762, but I am not so sure, I Will play with the hardware for understanding the precise SoC they are using.

atc1441, 2024-06-24:
Yeah also missed the detail that you where referring to the "R02_1.00.05_211220.bin" and not the latest "R02_3.00.06_240523.bin" which I used to reverse engineer. The firmware dump also includes that newer R02_3.00.06 FW: https://github.com/atc1441/ATC_RF03_Ring/tree/main/Firmware_Dump

So based on that completely different firmware with RTL there is a very high chance that there are indeed 2 Hardware versions of the R02 ring out there, (The other named rings like R03 etc. will most likely have the same hardware as R01 or R02)

Do you have a ring already and see the Chip inside?

decaduto, 2024-06-24:
> Yeah also missed the detail that you where referring to the "R02_1.00.05_211220.bin" and not the latest "R02_3.00.06_240523.bin" which I used to reverse engineer. The firmware dump also includes [...]

I have that ring physically, very soon I Will give you the chip name.

decaduto, 2024-06-24:
> Yeah also missed the detail that you where referring to the "R02_1.00.05_211220.bin" and not the latest "R02_3.00.06_240523.bin" which I used to reverse engineer. The firmware dump also includes [...]

The Amazon details for this smart ring are at: Fiorky R02 Smart Ring Fitness Tracker Ring 5.0 Tracker di ossigeno nel sangue compatibile con Bluetooth IP68 e modalità multi-sport imformeabili 5ATM for Android for IOS https://amzn.eu/d/0ea4cz5x

Note that Is cheaper than the One with rf03 SoC

atc1441, 2024-06-24:
Ok looking forward to it.

Unfortunately RTL is it pain in the as to work with!
Its normally on my Blacklist 😅 but at some point it needs to be understood anyway

decaduto, 2024-06-24:
> Ok looking forward to it.
> 
> Unfortunately RTL is it pain in the as to work with! Its normally on my Blacklist 😅 but at some point it needs to be understood anyway

I have worked with some RTL SoC for the k56 pro smartwatch, I agree, the SDK I linked above seems also incomplete, I am planning to buy Also the rf03 and the One based on the DA SoC on Amazon

atc1441, 2024-06-24:
What a bummer, this will make things very complicated. Since there is no way to tell them apart its pure luck which one someone will get.

So even if a custom firmware is available for both/all types there is much more to be checked, luckily they made the magic number different form RF03 to RTL so the firmware will not get accepted by each others Ring

decaduto, 2024-06-24:
> What a bummer, this will make things very complicated. Since there is no way to tell them apart its pure luck which one someone will get.
> 
> So even if a custom firmware is available for both/all types there is much more to be checked and to be explained to new users, luckily they made the magic number different form RF03 to RTL so the firmware will not get accepted by each others Ring

This Is interesting too, can we move the conversation on another social like telegram or discord? I think It should be better for improving the communication.

atc1441, 2024-06-24:
Sure, feel free to join my Discord: https://discord.gg/4vgFZgJY

There is channel for the Ring already :)

decaduto, 2024-06-24:
I cannot write on the discord group, any help?

kondas, 2024-09-13:
@atc1441 Can i ask for a discord invite? The link is not working above. Thanks in advance

kondas, 2024-09-27:
I've succesfully built the SDK3's "ble_adv_base" example using Keil. Unfortunately it needs a full license because it uses V5 compiler, so does not fit into Community limitations. It produced an ota bin, which i tried to flash via [https://atc1441.github.io/](https://atc1441.github.io/ATC_RF03_Writer.html) but after uploading it yields an error: DOMException: GATT operation failed for unknown reason. 
@decaduto You can obtain fromelf from [here](https://developer.arm.com/downloads/view/ACOMP5?revision=r5p6-07rel1) after registering an account.

decaduto, 2024-09-28:
> I've succesfully built the SDK3's "ble_adv_base" example using Keil. Unfortunately it needs a full license because it uses V5 compiler, so does not fit into Community limitations. It produced an ota bin, which i tried to flash via [https://atc1441.github.io/](https://atc1441.github.io/ATC_RF03_Writer.html) but after uploading it yields an error: DOMException: GATT operation failed for unknown reason. The start of the file looks different than the Colmi OTA firmware binary, maybe it is different OTA from SDK2, or custom implementation by Colmi.
> 
> @decaduto You can obtain fromelf from [here](https://developer.arm.com/downloads/view/ACOMP5?revision=r5p6-07rel1) after registering an account.

Extremely interesting, can we move to another chat? Like discord o telegram? You can join the discord group, I was working on this sdk too!

kondas, 2024-09-28:
Can You suggest a channel, or discord server? The link to @atc1441 discord invite is not working for me. I've also managed to build some stuff from SDK2, and the OTA binary is more similar to the Colmi one, but still different.

decaduto, 2024-09-28:
> Can You suggest a channel, or discord server? The link to @atc1441 discord invite is not working for me. I've also managed to build some stuff from SDK2, and the OTA binary is more similar to the Colmi one, but still different.

https://discord.gg/P23J9bnx

kondas, 2024-11-09:
@MenneBos https://discord.gg/ppEzgdmN

luisomoreau, 2024-11-19:
Hi guys,

I'm arriving a bit late to the party :) 

I tried to decompile the firmware with BinaryNinja to try to make sense of it.
I can see the firmware version mentioned in several part of the decompiled binary.

@atc1441, I'm interested in understanding how you found which part to update to create your FasterRawValuesMOD?

Best,

Louis

luisomoreau, 2024-11-21:
Sure, I'd be happy to help!

luisomoreau, 2024-11-22:
@decaduto, How do I join the discord?
I see that the links are not valid anymore

decaduto, 2024-11-22:
Sorry for the delay, here the link: https://discord.gg/h9DJuHqX

VedantParanjape, 2025-09-13:
@decaduto @atc1441 I also bought a R06 ring, and want to take a look at the decompiling. Could you add me to the discord chat too?