atc1441/ATC_RF03_Ring #5: R06 compatibility?  No size 12 for R02 is why.

AkitaAttribute, 2024-06-26:
I have fat fingers, and there weren't any options for size 12 on the R06.  Wondering if you can figure it out for the R06 :)

atc1441, 2024-06-26:
Very likely the R06 is the same hardware as R02. But just a gueess right now.

joetancy, 2024-06-27:
Yeah they are the same. I bought a R06 and was able to use the flasher to send commands

atc1441, 2024-06-27:
@joetancy did you also try a firmware update? Commands are the same even with different SoC Or are you maybe able to get a closeup picture of the Chip?

joetancy, 2024-06-28:
[https://ibb.co/album/93ygkj](https://ibb.co/album/93ygkj)
here you go, potato quality photos i'm afraid.

im gonna try firmware update next

atc1441, 2024-06-28:
Hey @joetancy  Yes that is definitely the same hardware :) 

Thank you

joetancy, 2024-06-28:
would try to flash a firmware but can't find any to do so, can you lead me to it?

atc1441, 2024-06-28:
This is the "normal" Stock firmware https://github.com/atc1441/ATC_RF03_Ring/blob/main/OTA_firmwares/R02_3.00.06_240523.bin

bitbooster, 2024-06-29:
From this post (https://www.reddit.com/r/SmartRings/comments/1d43edc/another_clone_colmi_r06/) that includes some notes from Colmi support, it seems that R02, R03 and R06 models all have the same chipset and set of sensors. They have only minor external differences (R03 has some 'lenses' on top of the sensors, and R06 a different design for the metal part)

otonieru, 2024-07-17:
> This is the "normal" Stock firmware https://github.com/atc1441/ATC_RF03_Ring/blob/main/OTA_firmwares/R02_3.00.06_240523.bin

Hello, may i ask, what is the changes in the FASTERRAWMOD firmware ? 

i have R06 as well, would it be possible to use the mod file on it, considering all hardwares are the same

otonieru, 2024-07-17:
> would try to flash a firmware but can't find any to do so, can you lead me to it?

Hello Joe, i see that you also have R06. Did you manage to update the firmware using any given files ? TIA