atc1441/ATC_RF03_Ring #7: Firmware difference

Stan2Be, 2024-07-06:
Hi!

Whats exactly the difference between the original OTA Firmware and the one you posted later:

R02_3.00.06_FasterRawValuesMOD.bin

atc1441, 2024-07-06:
This enables faster reading of RAW sensor values. 
See the https://atc1441.github.io/ATC_RF03_Writer.html tool and click on "Enable Animation" before and after to see the difference

Stan2Be, 2024-07-06:
thank you for your fast answer! 

Your E-Mail. Your Cloud. Your Office. eclipso Mail Europe. https://www.eclipso.de

Stan2Be, 2024-07-17:
Does this firmware fits to the R03 or R06 ring? Ir only for the R02?

otonieru, 2024-07-17:
> Does this firmware fits to the R03 or R06 ring? Ir only for the R02?

so i bite the bullet and tried to flash the firmware to my own R06. Happy to tell that it works just fine, aside that the device name now change to R02. But i think it can be modified in the .bin itself. Might try to change it myself later

cc : @atc1441

Stan2Be, 2024-07-17:
I tried to flash it to the R03 it wrote wrong CRC... do it doest work. But im also getting R06 soon wpuld ypu be so kind and upload the file for the R06 firmware too? I cant modify it myself... thank you

otonieru, 2024-07-17:
> I tried to flash it to the R03 it wrote wrong CRC... so it doest work. But im also getting R06 soon so would you be so kind and upload the file for the R06 firmware too? I cant modify it myself... thank you

I use the MOD firmware from here : https://github.com/atc1441/ATC_RF03_Ring/blob/main/R02_3.00.06_FasterRawValuesMOD.bin and it actually works

atc1441, 2024-07-17:
@Stan2Be  you can just flash it to R03 and R06 even while the Bluetooth name changes then to R03 for both its not an issue as the hardware is the same

Stan2Be, 2024-07-17:
I tried but it wrote CRC error and stopped

Stan2Be, 2024-07-17:
It worked with the modded firmware but now renamed the ring to R02 instead of R03…. How can I change it back?

Stan2Be, 2024-07-22:
Is there a way to rename the ring ahsin? Because i have another R02 and now both comes as R02...

adiif1, 2024-08-10:
@atc1441 Hi!, if possible to change read value in apk qring from 5 min to 1?  I use your mod software

Stan2Be, 2024-08-10:
i cant do such thing alone... im a newbie to this things :(

devnoname120, 2024-08-16:
@atc1441 How hard would it be to increase the frequency of automatic HR/SPO₂ measures beyond the 5 minute/1 hour treshold?

Stan2Be, 2024-08-16:
i just don't know how to make it... i just found your Guide and Thats how i Flash it but i ganz do more

adiif1, 2024-08-31:
> i just don't know how to make it... i just found your Guide and Thats how i Flash it but i can't do more... can you help me?
> […](#)
> --- Ursprüngliche Nachricht --- Von: Paul ***@***.***> Datum: 16.08.2024 18:04 An: atc1441/ATC_RF03_Ring ***@***.***> Betreff: Re: [atc1441/ATC_RF03_Ring] Firmware difference (Issue #7)
> ________________________________________________________ Your E-Mail. Your Cloud. Your Office. eclipso Mail Europe. https://www.eclipso.de

buddy but read from beginning to end.. see who we mark as a clue - the author...

Andre-Lehrmann, 2024-09-01:
Hi, i flashed a R06 Ring in 12 and in 11 with [R02_3.00.06_FasterRawValuesMOD.bin] (https://github.com/atc1441/ATC_RF03_Ring/blob/main/R02_3.00.06_FasterRawValuesMOD.bin) and both rings are working. 

Is it possible to change the firmware so that the ring measures every 6 minutes?

According to adverts, the ring is supposed to be able to store 7 days of data. That is 48 measurements per day times 7 = 336 measuring points.
If measurements are now taken every 6 minutes, then the time until the mobile phone has to read out the data at the latest would be: 6 * 336 =2016 minutes = approx. 33 hours. This would get you through the day without any problems and give you more accurate results.

Stan2Be, 2024-09-01:
I know i can flash it i just don’t want the name of the ring to be changed to R02 like it happened with my R03 ring…

atc1441, 2024-09-01:
> I know i can flash it i just don’t want the name of the ring to be changed to R02 like it happened with my R03 ring…

The name does not matter actually. But your decision of course :)

Stan2Be, 2024-09-01:
True! What’s why u asked if there is a possibility to change the name in the firmware to R06… but @atc1441  didn’t answered…

atc1441, 2024-09-01:
Yes it can be changed as shown in the Video :)
https://youtu.be/IOMqtFrNpTI

Stan2Be, 2024-09-01:
I just watched the video again but I’m totally newbie into this and I didn’t understood how to change the name of the ring when I flash the software… 😢

Andre-Lehrmann, 2024-09-01:
I know this is not your job @atc1441 , but could you please give me a tip on which tool I can use to change the firmware? 

I don't mean HeXd, as I can already change values there, 
but that would be rather random. Is there a tool that translates the firmware into any programming language (except assembler!), something like C# or another higher programming language? 

@StanBe - you have to find the right place where the firmware name is written. Change this value with the Hex tool and enter a new checksum at the top. Then you have patched it yourself.

Stan2Be, 2024-09-01:
That means I have to download something on my computer so I can see all this things? I used only the flash tool until now…

Andre-Lehrmann, 2024-09-01:
Yes, you need a Hex Editor, use the recommend in the video. And the firmware as a file.

Stan2Be, 2024-09-01:
Thank you I will give it a try

atc1441, 2024-09-01:
> I know this is not your job @atc1441 , but could you please give me a tip on which tool I can use to change the firmware?
>
> I don't mean HeXd, as I can already change values there, but that would be [...]

There is no tool like that.

You can use Ghidra to look into the firmware but edits need to be on hex level

Stan2Be, 2024-09-01:
@atc1441 i tried with this software but it didn’t worked :( is it a lot of work if you change it for me to R06v? Please 🙏