atc1441/ATC_RF03_Ring #8: Gadgetbridge

Izaic, 2024-07-06:
What would it require to get this ring working with an alternative app such as gadgetbridge? Does it use proprietary protocols?

octoTurnip, 2024-07-16:
I think this would be amazing if it can work! Have you seen any update to this?

devnoname120, 2024-07-25:
Someone will need to step in to implement this new device in GadgetBridge based on this information.

See this tutorial from GadgetBridge: https://gadgetbridge.org/internals/development/new-gadget

And here is a list of the available BLE commands: https://github.com/atc1441/ATC_RF03_Ring/issues/13

devnoname120, 2024-07-28:
Oh wow it looks like someone is already on it: https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/3896

derkrasseleo, 2024-07-30:
> Oh wow it looks like someone is already on it: https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/3896

But that's for the default firmware, right? Or would that work with this firmware too?

Nismonx, 2024-07-30:
> > Oh wow it looks like someone is already on it: https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/3896
> 
> But that's for the default firmware, right? Or would that work with this firmware too?

Both, assuming when you say "this firmware" you [this?](https://github.com/atc1441/ATC_RF03_Ring/blob/main/R02_3.00.06_FasterRawValuesMOD.bin)