atc1441/ATC_RF03_Ring #10: A few more commands to send to the ring

devsnd, 2024-07-13:
Hey @atc1441,

Thanks a lot for your efforts! You motivated me to get one of these rings myself, because I wanted to do some sleep tracking. I looked at the qring app and found probably most the commands you can send to the ring. Here's a quick list:

```
BindAncsReq
402000000000000
blackListReq
2D01000000000000
deviceAvatarReq
3200000000000000
findDeviceReq
5055AA0000000000
setANCSReq
60FF9F0000000000
bpReadConformReq
E00000000000000
cameraReq
205000000000000
agpsReq
3001000000000000
bloodOxygenSettingReq
2C01000000000000
bpSettingReq
C01000000000000
brightnessSettingsReq
1B01000000000000
degreeSwitchReq
1901000000000000
displayClockReq
1201000000000000
displayOrientationReq
2901000000000000
displayStyleReq
2A01000000000000
displayTimeReq
1F01000000000000
dndReq
601000000000000
heartRateSettingReq
1601000000000000
intellReq
901000000000000
pressureSettingReq
3601000000000000
readPersonalizationSettingReq
1701020300000000
startHeartRateReq_ACTION_CONTINUE
6903250000000000
startHeartRateReq_ACTION_PAUSE
6902000000000000
startHeartRateReq_ACTION_START
6901000000000000
startHeartRateReq_ACTION_STOP
6904250000000000
startHeartRateReq_TYPE_BLOODOXYGEN
6903250000000000
startHeartRateReq_TYPE_BLOODPRESSURE
6902000000000000
startHeartRateReq_TYPE_BLOOD_SUGAR
6909250000000000
startHeartRateReq_TYPE_ECG
6907250000000000
startHeartRateReq_TYPE_FATIGUE
6904250000000000
startHeartRateReq_TYPE_HEALTHCHECK
6905250000000000
startHeartRateReq_TYPE_HEARTRATE
6901000000000000
startHeartRateReq_TYPE_PRESSURE
6908250000000000
startHeartRateReq_TYPE_REALTIMEHEARTRATE
6906250000000000
stopBloodOxygen
6A03000000000000
stopHeartRate
6A01000000000000
stopBloodPressure
6A02000000000000
stopFatigue
6A04000000000000
stopHealthCheck
6A05000000000000
stopPressure
6A08000000000000
stopBloodSugar
6A09000000000000
readPressureReq
1400000000003200
readSleepDetailsReq
44010F005F000000
readTotalSportDataReq
701000000000000
restoreKeyReq
FFFFFF0000000000
setTimeReq
124071311004601
```

For example, when I send the blood oxygen command (`690325`), I get `10:38:32 : Heart Rate at: 98 bpm`, which is probably 98% blood oxygen.

I just wanted to send you this, so you know there is interest in your project :)

Sakd99, 2024-07-17:
@devsnd @atc1441 

Can I have a copy of this web project please

https://atc1441.github.io/ATC_RF03_Writer.html

I need to convert it into Arabic for my graduation project

tahnok, 2024-07-28:
Having looked through a lot of the code, I feel like many of these are leftover commands from a smart watch, but it's hard to tell. Like I don't think AGPS, [ACNS](https://developer.apple.com/library/archive/documentation/CoreBluetooth/Reference/AppleNotificationCenterServiceSpecification/Specification/Specification.html) or displayOrientationReq make any sense in the context of this ring

That said I can confirm that the blood oxygen command matches what I uncovered before finding this issue

devsnd, 2024-07-29:
I've worked a bit more on the reverse engineering here. It's true that many commands do not work for the ring, but some other smart devices that are also compatible with the qring app.

I am now able to generate a bunch more commands with different parameters to set things like heart-rate measurement interval and will try to package all of it up into some javascript functions that can be used as a starting point in the SPA provided by ATC. (https://atc1441.github.io/ATC_RF03_Writer.html)