atc1441/ATC_RF03_Ring #13: BLE commands and services

devnoname120, 2024-07-25:
I did some reverse engineering and here are my findings:

**List of services:**

```c
UUID_SERVICE = "6e40fff0-b5a3-f393-e0a9-e50e24dcca9e"
UUID_READ = "6e400003-b5a3-f393-e0a9-e50e24dcca9e"
UUID_WRITE = "6e400002-b5a3-f393-e0a9-e50e24dcca9e"
GATT_NOTIFY_CONFIG = "00002902-0000-1000-8000-00805F9B34FB"
SERVICE_DEVICE_INFO = "0000180A-0000-1000-8000-00805F9B34FB"
CHAR_FIRMWARE_REVISION = "00002A26-0000-1000-8000-00805F9B34FB"
CHAR_HW_REVISION = "00002A27-0000-1000-8000-00805F9B34FB"
CHAR_SOFTWARE_REVISION = "00002A28-0000-1000-8000-00805F9B34FB"
SERIAL_PORT_SERVICE = "de5bf728-d711-4e47-af26-65e3012a5dc7"
SERIAL_PORT_CHARACTER_NOTIFY = "de5bf729-d711-4e47-af26-65e3012a5dc7"
SERIAL_PORT_CHARACTER_WRITE = "de5bf72a-d711-4e47-af26-65e3012a5dc7"
```

**List of commands:**

```c
CMD_SET_DEVICE_TIME = 1;
CMD_TAKING_PICTURE = 2;
CMD_GET_DEVICE_ELECTRICITY_VALUE = 3;
CMD_SET_PHONE_OS = 4;
CMD_FANWAN = 5;
CMD_MUTE = 6;
CMD_GET_STEP_TOTAL_SOMEDAY = 7;
CMD_RE_BOOT = 8;
CMD_INTELL = 9;
CMD_GET_TIME_SETTING = 10;
CMD_BP_TIMING_MONITOR_SWITCH = 12;
CMD_BP_TIMING_MONITOR_DATA = 13;
CMD_HR_TIMING_MONITOR_DATA = 13;
CMD_BP_TIMING_MONITOR_CONFIRM = 14;
CMD_HR_TIMING_MONITOR_CONFIRM = 14;
CMD_BIND_SUCCESS = 16;
CMD_PHONE_NOTIFY = 17;
CMD_DISPLAY_CLOCK = 18;
CMD_GET_SPORT = 19;
CMD_GET_BAND_PRESSURE = 20;
CMD_GET_HEART_RATE = 21;
CMD_HR_TIMING_MONITOR_SWITCH = 22;
CMD_GET_PERSONALIZATION_SETTING = 23;
CMD_GET_DEGREE_SWITCH = 25;
CMD_SEND_WEATHER_FORECAST = 26;
CMD_GET_BRIGHTNESS = 27;
CMD_GET_MUSIC_SWITCH = 28;
CMD_MUSIC_COMMAND = 29;
CMD_REAL_TIME_HEART_RATE = 30;
CMD_DISPLAY_TIME = 31;
CMD_CALIBRATION_RATE = 32;
CMD_TARGET_SETTING = 33;
CMD_FIND_THE_PHONE = 34;
CMD_SET_ALARM_CLOCK = 35;
CMD_GET_ALARM_CLOCK = 36;
CMD_SET_SIT_LONG = 37;
CMD_GET_SIT_LONG = 38;
CMD_SET_DRINK_TIME = 39;
CMD_GET_DRINK_TIME = 40;
CMD_ORIENTATION = 41;
CMD_DISPLAY_STYLE = 42;
CMD_MENSTRUATION = 43;
CMD_AUTO_BLOOD_OXYGEN = 44;
CMD_BLACKLIST_LOCATION = 45;
CMD_PACKAGE_LENGTH = 47;
CMD_AGPS_SWITCH = 48;
CMD_DEVICE_AVATAR = 50;
CMD_PRESSURE_SETTING = 54;
CMD_PRESSURE = 55;
CMD_HRV_ENABLE = 56;
CMD_HRV = 57;
CMD_GET_STEP_SOMEDAY_DETAIL = 67;
CMD_GET_SLEEP = 68;
CMD_QUERY_DATA_DISTRIBUTION = 70;
CMD_GET_STEP_TODAY = 72;
CMD_ANTI_LOST_RATE = 80;
CMD_GPS_ONLINE = 84;
CMD_SET_ANCS_ON_OFF = 96;
CMD_GET_ANCS_ON_OFF = 97;
CMD_START_HEART_RATE = 105;
CMD_STOP_HEART_RATE = 106;
CMD_HEALTH_ECG_START = 108;
CMD_HEALTH_ECG_DATA = 109;
CMD_HEALTH_PPG_DATA = 110;
CMD_ECG_STATUS_DATA = 111;
CMD_ECG_MEASURE_TIME = 112;
CMD_MSG_NOTIFY = 114;
CMD_PUSH_MSG = 114;
CMD_DEVICE_NOTIFY = 115;
CMD_TUNE_TIME_DIRECT = 115;
CMD_PHONE_GPS = 116;
CMD_TUNE_TIME_INVERSE = 116;
CMD_PHONE_SPORT = 119;
CMD_PHONE_SPORT_N0TIFY = 120;
CMD_MSG_GET_HW_AND_FW_VERSION = 147;
CMD_TEST_OPEN = 201;
CMD_TEST_CLOSE = 202;
CMD_RE_STORE = 255;
```

atc1441, 2024-07-25:
Hey, these CMDs where already reverse engineered and some are implemented in the WebFlasher for simple use :)

https://atc1441.github.io/ATC_RF03_Writer.html

devnoname120, 2024-07-25:
@atc1441 Are they documented somewhere?

atc1441, 2024-07-25:
The source from the https://atc1441.github.io/ATC_RF03_Writer.html is pretty good readable for that matter :)

devnoname120, 2024-07-25:
@atc1441 I only see a few commands documented in `checkNotifyData()` in the source code. Did I miss something?