atc1441/ATC_RF03_Ring #16: wrong crc: Colmi R03

balaclava-guy, 2024-09-23:
Hello, I'm trying to use the web flasher to update a Colmi R03 using the binary file here: 

https://github.com/atc1441/ATC_RF03_Ring/blob/main/R02_3.00.06_FasterRawValuesMOD.bin 

#7 suggests it should be possible to flash an R03 with the R02 firmware. 

I've re-downloaded the file several times now with the same result.

I have tried on both a mobile (Galaxy S8) and a Macbook Pro (via Chrome) - same result.

```
16:23:17 : Connecting to: R03_2305
16:23:20 : > Found write characteristic
16:23:21 : Found custom Main service
16:23:21 : Found custom write characteristic
16:23:21 : Found custom notify characteristic
16:23:21 : CONNECTED, you can now select the firmware and it will be directly flashed.
16:23:21 : Send CMD: 03000000000000000000000000000003

16:23:44 : Length OTA: 0ccb0100
16:23:44 : CRC All: 9a56
16:23:44 : Checksum All: dc4f
16:23:44 : Send CMD: bc010000ffff
16:23:44 : Got bytes: bc010100bf4000
16:23:44 : Send CMD: bc020900abd4010ccb01009a56dc4f
16:23:44 : Got bytes: bc020100bf4000
16:23:44 : Sending OTA Part: 0 from: 115
16:23:45 : Got bytes: bc0301003e8102
16:23:45 : Rx wrong CRC!
```

nikbasi, 2024-09-23:
I have the same issue. Mine also came with FW 3.00.10 and is recognised as R03_5306.

atc1441, 2024-09-23:
sounds like they changed something after my tinkering.

atc1441, 2024-09-23:
Ok so the reason is that the devices you got are a different hardware type.

They do not use the BlueX SoC. 

Please share an in depth picture of the SoC in the Epoxy to verify it.

The CRC check has a different start value on that type to prevent wrong flashing, which does work :)

atc1441, 2024-09-23:
Also please use the nRFConnect App to read the Device information's like in this image to verify it more.

[image]

AnalyticETH, 2024-11-01:
Hey @atc1441  what is the CRC algorithm? Might try to update this to 3.00.10 this weekend