atc1441/ATC_RF03_Ring #19: [Question] Using custom Firmware, could I use this as a HR Monitor ?

NiyahVE, 2024-10-17:
Hello everyone.

I stumbled upon Colmi and all those custom firmware recently and it got me asking myself question:

Could I use this as a heart rate monitor without using the provided apps ?

What I'm trying to do / reach here is the ability to extract my heart rate way faster than every 30 or 5 minutes and put the value in a .txt.

I saw the custom firmware here: [R02_3.00.06_FasterRawValuesMOD.bin](https://github.com/atc1441/ATC_RF03_Ring/blob/main/R02_3.00.06_FasterRawValuesMOD.bin)

but how "Faster" are we talking about ?  I don't care about oxygen and all that, I'm just looking to update my heart rate every 3 to 5 seconds and put the value in a .txt file on my computer, does that seem like a doable thing ? 

Would the battery last at least some hours by doing so ? (I'm looking for 6-8 hours battery lifetime at best).

I'm mainly asking for people that have one of those Colmi ring already since I'm still debating on getting one, but the project look like a fun thing to do.

Kdcius, 2024-11-06:
Hi dude !
Sorry no answer (yet), however that is exactly the goal I'm trying to acheive.
So just commenting to get any update on here, and I'll keep you in touch if I did manage to get this working ;)

NiyahVE, 2024-11-06:
Good to know I'm not the only one looking for this, please keep me updated ! :)

Kdcius, 2024-11-07:
Okay after reading trough all this gh and related ones, I think I got this.

You don't need a CFW, or maybe if you want a faster rate. But with the stock firmware you can have a refresh rate of approx 1hz, so probably enough, you can try flashing the custom firmware that is supposed to increase that, I cannot tell you how much it will increase but I guess it should be more than 1hz !

So if you need to log the raw data you can either use [this](https://github.com/edgeimpulse/example-data-collection-colmi-r02?tab=readme-ov-file) which sound promising. Or inspect the page of the [tool](https://atc1441.github.io/ATC_RF03_Writer.html) made by @atc1441 (thanks you so much by the way !) to create your own app (python, js, or whatever you feel like) to parse the data by using the same kind of functions.

I think tomorrow I'll try the existing script (first link). Then create my own python script as I want to pass some of theses precious bio metrics to a neural net hehe.

NiyahVE, 2024-11-07:
So the stock firmware have a refresh rate of about 1hz, which mean 1 cycle per seconds but it default to taking the datas every 5 secs or so ? That's interesting, and also mean we can totally skip the mod then.

I'll get myself a ring soon to test stuff out on my side as well then :)

Thank you !

Kdcius, 2024-11-07:
I think not, it seem to refresh every seconds, but you can get it to refresh faster with the mod.

Nice, go ahead ! I got one and was expecting to have to sniff the BLE packets to reverse engineer the protocol myself.
I really didn't expected to find such a community supporting this little ring ! That's really cool !

Kdcius, 2024-11-07:
Well, I was wrong actually ^^'
You can get the raw data, but it's not the processed heart rate metrics. It's the sensor raw data that you have to use to calculate the user's heart rate (BPM).
Even with the custom firmware (which I tried installing on my R06), you only get updates on the BLE at around 2-3Hz, and that's not enough to calculate heart rate (it needs at least 20Hz). So, the only way I've found for now is to send the command 690101 = Measure Heart Rate - that tells the ring to measure the HR, process the data in the ring, and then send back the HR value after about 20 seconds.
So the best you can probably get is maybe 2 or 3 measurements per minute with the ring without modifying the firmware even more than the actual custom firmware ATC made.
I'm still trying to find a way to maybe start a continuous measurement and HR calculation that streams some kind of live processed result.