atc1441/ATC_RF03_Ring #22: Works with the brand-new R10!

andiohn, 2024-11-03:
So far I can connect and read the heartrate etc. Kinda cool that this also supports that ring.

girlmaya, 2025-01-06:
Where did you buy it from? If aliexpress, could you send the url of the listing?
The colmi website says the R10 has a RTL8762 ESF, which would not be supported by this custom firmware:
https://www.colmi.com/colmi-r10-smart-ring/

andiohn, 2025-01-07:
Hey sorry, you're right. Somehow this post was posted to the alternative firmware here and not to the GadgetBridge repo, my apologies.

https://www.aliexpress.com/item/1005007765211101.html

JFingerle, 2025-05-03:
> Hey sorry, you're right. Somehow this post was posted to the alternative firmware here and not to the GadgetBridge repo, my apologies.
> 
> https://www.aliexpress.com/item/1005007765211101.html

Then close this issue?!

closed 2025-05-03 by atc1441