atc1441/ATC_RF03_Ring #23: Tap Detection Sensitivity too high

Kdcius, 2024-11-07:
## Current Setup
I'm using the Ring as a tap detection device to control smart home appliances with the following setup:
- Android app with a background service
- Ring configured in "Shoot Camera motion feedback" mode for tap detection
- MQTT integration with HomeAssistant server

## Issue
The tap detection sensitivity is too high, causing unwanted triggers when walking around the house.

## Attempted Solutions
1. **Firmware Analysis**
   - Searched for accelerometer address in the custom firmware binary
   - Attempted to decompile firmware with Ghidra (beyond current expertise)

2. **Software Workarounds**
   - Implemented multiple tap confirmation
   - Issue persists: sometimes triggers multiple times involuntarily

## Additional Information
The system works almost flawlessly except for the sensitivity issue.

## Request
Looking for recommendations to reduce tap detection sensitivity or alternative approaches to prevent false triggers.

richrd, 2024-11-10:
I have been playing around with the tap detection but for me it's not very responsive and misses half my taps. I haven't had it enabled during day to day activities so can't comment on that. Modifying the firmware is probably the only way to improve it unless there are some undocumented sensitivity settings we can change on the ring somehow.

I'm also interested in doing something with the ring and home assistant. What is the range like or do you have Bluetooth repeaters? Also did you write the custom Android app?

I'm working on a webapp to replace the shady Chinese app and will hopefully be ready to publish something soon.

Kdcius, 2024-11-11:
Yeah it is not responsive; in fact, it will ignore all taps 2 seconds after sending the first notification, so I cannot even try to detect double or triple tap...

I managed to have around 3 meters of range away from my phone. I'm not using any repeaters, the plan was to use the phone as the relay to HA because I always have it near me.

And yes I did write a native Android app in Android Studio. It has a service that keeps running in background or when the phone is locked. The service is continuously watching for notifications from the ring, and it will publish a message to my HA server via MQTT.

I could share with you the source if that interests you. But it's my first native app ever so it is a bit messy ^^'

Also, if you haven't already, take a look at [Gadgetbridge](https://gadgetbridge.org/) app, someone has integrated the Colmi rings series in it. So it can basically replace all the functions of the Chinese app!

richrd, 2024-11-12:
Nice! I've not done much native Android development and don't have Android Studio at the moment so the source might not be of use to me right now but it's always interesting to take a look if you feel like publishing it somewhere. Might open the door to making some of my own utility apps 🤔 

I wasn't aware that Gadgetbridge had support for these rings, awesome! Thanks for the heads up :) Looks like it's all I need, but I might still have some fun building my webapp anyway.