atc1441/ATC_RF03_Ring #25: New Product?

FreeMasen, 2024-12-09:
When looking at the Colmi R0X devices on AliExpress, I noticed a new product recomendation that _looks_ a lot like these rings but provides a screen and touch controls. Not sure if you've seen this or not but I couldn't find much about it when searching.

https://www.aliexpress.us/item/3256807937890584.html

It might fill in the gaps you outlined on youtube about it being under capable for further development

atc1441, 2024-12-10:
Thank you. Ordered one, the SoC is the Dialog DA14585 so not compatible with the other rings and maybe even signed OTA updates which would be bad

girlmaya, 2025-01-06:
@atc1441 Did you take a look at how the firmware updates work?

girlmaya, 2025-01-07:
Here's another interesting ring, this one has NFC as well as health data. No idea if the sensors are any good, but they openly state the chip name "FR8016HA" in the listing. Data sheet here: https://www.lcsc.com/datasheet/lcsc_datasheet_2312081753_FREQCHIP-FR8016HA_C2688843.pdf

FreeMasen, 2025-03-13:
https://github.com/atc1441/ATC_SR08_Ring

closed 2025-03-13 by FreeMasen