atc1441/ATC_RF03_Ring #26: R02 ring with a different bluetooth chip - 8762ESF

adokitkat, 2024-12-21:
Hello.

I got COLMi R02 ring and wanted to flash the firmware from this repo, however I got invalid CRC error. After looking around in issues here and then looking on the chip in the ring I found out it seemingly is ["BM-8762ESF" chip](https://www.realmcu.com/img/partner/en_638367902219467641.pdf).

Is it possible to modify firmware for this version as well please?

[image]
[image]
[image]
[image]

adokitkat, 2024-12-22:
This should be the latest firmware .bin I found in app cache:

[RY02_3.00.31_241207.bin.zip](https://github.com/user-attachments/files/18220351/RY02_3.00.31_241207.bin.zip)

bbetter173, 2025-02-01:
I've bought a ring with the same chipset in it, after doing some research it supports all the usual stuff like encryption and readback protection with no known bypass (unless @atc1441 has another glitcher up his sleeve).

After doing some reading, the only real chance is if the manufacturer chose to leave it open same as they have for the RF03 rings. I'm going to try and dump the firmware when I get it, though even if I can I doubt reverse engineering the OTA update method will be something I can do.

djoek, 2025-02-24:
found another one... this was sold as a ColMi r10 but seems to have another RY firmware:
device info {'hw_version': 'RY03CR_V1.0', 'fw_version': 'RY03CR_1.00.06_241212'}
battery: BatteryInfo(battery_level=97, charging=False)

awra2001, 2025-03-15:
> found another one... this was sold as a ColMi r10 but seems to have another RY firmware: device info {'hw_version': 'RY03CR_V1.0', 'fw_version': 'RY03CR_1.00.06_241212'} battery: BatteryInfo(battery_level=97, charging=False)

I'm so glad I found this.  I'd been scratching my head trying to figure out what was happening with the R10 (I have one), but it looks like it's just a separate firmware issue.  I appreciate all the work atc1441 has done, but if there's something that could be done to help, I'm all about doing what I can.