atc1441/ATC_RF03_Ring #28: Invalid CRC error

Javitax47, 2025-01-03:
First of all, thank you for enlightening all of us with your hacking wisdom. Thanks to you I have been learning lots of things by giving a shot a project involving these rings. I appreciate the information and inspiration you provided by creating this repository/videos.

I got myself 2 Colmi R03 a few weeks ago and right away I tried uploading the fasterRAWvaluesMOD firmware provided in the repository and it worked fine. Now I tried again and got the invalid CRC error, which is weird because I read here that this is caused by having a different SoC on the ring, which obviously hasn't changed. Also, the firmware that is currently installed and I successfully tried to reinstall using the OTA flashing tool is **RY03_3.00.24_241227**. My objective is an even fasterRAWvaluesMOD that works on my ring, for that I have been studying the firmware using Ghidra but have not found a way to increase the raw data rate (#24 ). I also compared the fasterRAWvaluesMOD with the original version **R02_3.00.06_240523.bin** and found little to no change in the firmware memory (different CRC32 and a single byte that doesn't seem to matter). Can you explain how you modified the original firmware to increase the rate of RAW values so I can do it myself? Thanks in advance.

[image]

atc1441, 2025-01-03:
Hey, 
Yeah its unfortunate there is another SoC but the same named Product :-/

That one byte different in the faster firmware does make the difference :) its a timer value that is shorter after modifying it. It is the result of painstakingly reverse engineering the firmware and find the right byte to change

Javitax47, 2025-01-03:
Thanks for the confirmation! I knew that single byte was up to something... The firmware of the new version I have installed (RY03_3.00.24_241227) is quite different in structure, so I'll look for the byte that serves the same purpose and debug from there. I think I'll be closing the issue now.

closed 2025-01-03 by Javitax47

atc1441, 2025-01-03:
The RY03 firmware is completely different, keep that in mind. Different SoC different handling.
Have fun reverse engineering it :)
Im moved on to a different ring which will be released in a while