atc1441/ATC_RF03_Ring #30: Firmware Modded with RX wrong crc

antoniodangelophoto, 2025-03-21:
Hi, I'm trying to load the mod firmware to have a very high refresh rate for the raw ppg data, but during the loading I have a problem on the crc that seems corrupted, do you have any ideas? Thanks in advance

The message is this:
16:37:07 : Length OTA: 0ccb0100
16:37:07 : CRC All: 9a56
16:37:07 : Checksum All: dc4f
16:37:07 : Send CMD: bc010000ffff
16:37:07 : Got bytes: bc010100bf4000
16:37:07 : Send CMD: bc020900abd4010ccb01009a56dc4f
16:37:08 : Got bytes: bc020100bf4000
16:37:08 : Sending OTA Part: 0 from: 115
16:37:09 : Got bytes: bc0301003e8102
16:37:09 : Rx wrong CRC!

atc1441, 2025-03-21:
You have a different Ring it is not competible

atc1441, 2025-03-21:
It should be no problem then, can you read SoC name on the ring to verify

antoniodangelophoto, 2025-03-21:
The soc is 8762ESF

atc1441, 2025-03-21:
As expected it is the wrong SoC :) it is a different ring internally and not compatible

atc1441, 2025-03-21:
As the project name hints on the RF03 SoC is needed :)

atc1441, 2025-03-21:
I do not have this other ring so I can not add it, sorry

atc1441, 2025-03-21:
Just removed that firmware to not confuse again in the future :)

antoniodangelophoto, 2025-03-28:
sorry again, do you know which model on sale still contains the soc rf03?

nwrkbiz, 2025-08-03:
Interesting how many diffrent socs they use for these devices. I have one that looks exactly the same but is powered by a Dialog DA14531 (they even were kind enough to mentioned it on the package it came in ☺️).